TeamOpenFIRE/OpenFIRE-App #7: Nix package

keenanweaver, 2024-06-26:
Howdy! Saw your video about the project and was inspired to package it up for Nix and NixOS. I'll be glad to maintain the package. I'll probably also wait for a full non-RC release to submit it to the nixpkgs repo. I can automate the dialout group setting, but that requires a Nix module which would need some more research and effort.

If you wanted to include install instructions in the readme, the user can do several things:

Run temporarily with nix-shell or nix run:
`nix-shell -p openfire-app --run 'OpenFIREapp'`
or
`nix run nixpkgs#openfire-app -- 'OpenFIREapp'`

**Install to system or user packages via configuration.nix** (NixOS only)
```nix
environment.systemPackages = with pkgs; [ openfire-app ];
```
``` nix
users.users.yourUserName.packages = with pkgs; [ openfire-app ];
```
**Install to home via home.nix** (Any OS, requires nix & home-manager)
``` nix
home.packages = with pkgs; [ openfire-app ];
```

Here's the actual package I'll be submitting. Builds from source, creates the desktop item. Mostly used the AUR package as a guide.

``` nix
{
  lib,
  stdenv,
  fetchFromGitHub,
  cmake,
  qtbase,
  qttools,
  qtserialport,
  qtsvg,
  icu,
  copyDesktopItems,
  makeDesktopItem,
  wrapQtAppsHook,
}:

stdenv.mkDerivation (finalAttrs: {
  pname = "openfire-app";
  version = "2.0-rc3";

  src = fetchFromGitHub {
    owner = "TeamOpenFIRE";
    repo = "OpenFIRE-App";
    rev = "v${finalAttrs.version}";
    hash = "sha256-uq4aXA6YFsKRcByy5LqJPtPnImC0nx0iRHKM3BSbOAU=";
  };

  nativeBuildInputs = [
    cmake
    qtbase
    qttools
    qtserialport
    qtsvg
    icu
    copyDesktopItems
    wrapQtAppsHook
  ];

  desktopItems = [
    (makeDesktopItem {
      name = "OpenFIREapp";
      exec = "OpenFIREapp";
      icon = "org.TeamOpenFIRE.OpenFIREapp";
      desktopName = "OpenFIRE App";
      categories = [ "Utility" ];
      comment = "Configuration utility for the OpenFIRE lightgun system";
    })
  ];

  postInstall = ''
    mkdir -p $out/share/icons/hicolor/scalable/apps
    cp $src/ico/openfire.svg $out/share/icons/hicolor/scalable/apps/org.TeamOpenFIRE.OpenFIREapp.svg
  '';

  meta = {
    description = "Reference configuration/testing app for the OpenFIRE project, built in Qt/C";
    homepage = "https://github.com/TeamOpenFIRE/OpenFIRE-App";
    license = lib.licenses.gpl3Only;
    longDescription = ''
      In order to use the program successfully, add your username to the 'dialout' group with:
      ```nix
      users.users.yourUserName.extraGroups = [ "dialout" ];
      ```
    '';
    maintainers = with lib.maintainers; [ keenanweaver ];
    mainProgram = "OpenFIREapp";
    platforms = lib.platforms.linux;
  };
})
```

Now I have to look into getting a lightgun!

SeongGino, 2024-06-26:
Hiya, and thanks for the support!

I certainly welcome third party packaging; but I don't personally use nix or even know how it works, which is the only thing keeping me from being more enthusiastic or offering any more assistance.

But any more OpenFIRE exposure among the greater Linux certainly gets my full blessing. :)

SeongGino, 2025-04-11:
Hiya! Figured I should probably ask to see how this is going along, if at all?

I'm trying to keep the Issues page clean of stale tickets, and this happens to be the only one left at the moment. I'm also not sure if any of the updates made since the making of this thread has made the task easier or not.

keenanweaver, 2025-04-11:
Sounds good. I'll update whenever I get around to submitting it. I simply don't have the hardware to test if the package works quite yet.

closed 2025-04-11 by keenanweaver