TeamOpenFIRE/OpenFIRE-App #15: [Request] Ability to apply saved Custom Pin Mappings to other similar boards

SeongGino, 2024-08-25:
### Discussed in https://github.com/TeamOpenFIRE/OpenFIRE-App/discussions/14

<div type='discussions-op-text'>

<sup>Originally posted by **l3giona1re** August 22, 2024</sup>
**Request**: Save custom pin mapping and have option to upload this to a new board that has the same physical pin layout.

**Background:** Right now the app lets me save the custom pin mapping for a board, however, if I had to reset that board for what ever reason, when I load the firmware back on, I have to do the custom pin mapping all over again.
Also when I'm making multiple guns and want the same custom mapping, I would have to screencapture the pin mapping, and have that screencapture image next to OF app to replicate.</div>

SeongGino, 2024-08-25:
An initial version of this feature would likely to have to apply to exact same boards, though down the line this can be extended to "pin compatible" boards (though atm this would only apply to the Pico and Pico W).

Would the App need to create its own directory for default storage? Presumably it would have to be in `%APPDATA%/OpenFIRE` for Wendies, and `~/.config/OpenFIRE` for Linux et al

l3giona1re, 2024-08-25:
At the moment, the app is portable so I would suggest keeping with that theme so the mappings are saved to the OF app folder.
This would allow easier movement of the mapping between different PCs, e.g. when OF is run from a USB stick.

SeongGino, 2025-02-17:
Import/Export custom pinouts is waiting in #32, will close this issue when it's pulled.

closed 2025-03-05 by SeongGino