TeamOpenFIRE/OpenFIRE-App #18: [REQUEST] be able to select any neopixels in the series for a static colour

l3giona1re, 2024-09-09:
Currently you can only select first 3 neopixels for static colour configuration, leaving the rest to be configurable my things like Qmamehooker.

Sometimes for wiring purposes, it's cleaner to have the first neopixel placed where you want mamehooker to control and subsequent pixels, to be static.

Currently can't do this.

Request here is more granular control and static colour assignment of any neopixels wired.

SeongGino, 2024-09-09:
> here is more granular control and static colour assignment of any neopixels wired.

This isn't a pull request, not sure what the wording's for.

To allow *any* Pixels to be set as static or not wouldn't be practical, as it overcomplicates the way LED updates works.

The most that could be reasonably offered is a toggle determining whether static pixels are set from the first or the last X pixels of the strand; this requires work in the firmware to support the option first.

SeongGino, 2024-09-09:
Adding depends on TeamOpenFIRE/OpenFIRE-Firmware#26

l3giona1re, 2024-09-09:
I could work with that. Also could you allow more than just 3 static configuration?

SeongGino, 2024-09-09:
This issue is not for piling on other problems unrelated to the title of the issue you're posting in; all unique issues need separate posts please.

Regardless, adding more than three static emitters are not likely, because four pixels is close to if not the practical limit of how many emitters you can have running off USB power anyways.

SeongGino, 2025-04-03:
#43 (and TeamOpenFIRE/OpenFIRE-Firmware#64) addresses this issue by allowing the user to invert fixed color pixels to the last X pixels rather than only the first X, so for now I'll close this.

Perhaps at some point in the future, an alternative would be to have *one* static color, but allow the user to set which pixels out of the strand can be used (would only require an additional setting and a small UI modification) - that, or move custom NeoPixel settings to a separate preferences file and have a vector dedicated to custom LED colors (would require more breaking changes).

closed 2025-04-03 by SeongGino