TeamOpenFIRE/OpenFIRE-App #26: [REQUEST] Change test screen colors

Arakon, 2024-11-03:
I'm aware this is a fairly minor issue for most people, but in my case I just spent 30 mins trying to figure out why none of the buttons in the test button screen worked, until my wife pointed out that they do. I'm partially color blind and it's nearly impossible for me to see the thin text changing to red when a button is pressed. If a more prominent color could be used here, or the background of the field changed instead of the text, that would be a great help.

SeongGino, 2024-11-03:
Apologies for your visual impairment.

How does this look for you?
Dark mode on Linux: https://github.com/user-attachments/assets/020548e6-ae33-4f9b-a517-056e3716a608

Light mode on Windows 10: https://github.com/user-attachments/assets/215b986b-5589-4bde-8222-5479331134cb

Arakon, 2024-11-03:
That works great for me, very clear indication that the button is being pressed. Thanks for looking into this so quickly!

PR #27 (2024-11-03): Tests Tab: Use more prominent coloring & use stylesheets where applicable
Title. Also cleans up some code and comments.

Resolves #26
commit ab8daa5d (2024-11-03): Tests Tab: Use more prominent coloring & use stylesheets where applicable (#27)

Also cleans up some code and comments.

Resolves #26
closed 2024-11-03 by SeongGino