TeamOpenFIRE/OpenFIRE-App #56: Adding features for better management of the "Board Layout" screen

alessandro-satanassi, 2025-04-16:
It is not an important or essential thing, and it would only be useful for the "Board Layout" screen, to customize any other micros in addition to the rp2040 (e.g. Esp32). In rp2040 the analog pins are only from 26 onwards and the program controls these numbers. In rp2040 the pins that can be used as i2c wires must be in certain modes and the program controls this. In esp32 instead, for example, almost all the pins can be configured as analog, as well as ic2 without forcing between wire and wire1. It would be useful to add perhaps to the 'board' submodule a constant #define, which identifies the type of micro (e.g. rp2040, esp32, etc.) so that the OpenFire App can in the future adapt the "Board Layout" screen based on the possibilities of the micros. Or you could always define in Board a map of the pins of the micro where it is indicated if it can be used as analog, i2c0 or i2c1, or similar), I don't know, as you think is best. All this would only serve for a more correct display of the "Board Layout" and to 'use Custom Pins', or disable it in the case in which it is not an rp2040 platform, for example (therefore allowing customization only to the rp2040).

SeongGino, 2025-04-16:
I will say that it's *possible,* and defo needed, but it would upend some of the routines regarding how I2C channel conflicts are handled and how the layout is actually initialized.

A lot of it works under the assumptions of the RP2040 environment (naturally), so it's always assumed that the upper four pins are always analog, and that I2C pairs are always grouped in SDA-then-SCL on alternating blocks for every pair; the former would have to be changed *anyways* if we want to support RP2350**B**-based boards (not that I expect anyone to use them, but just covering my ass here).

I don't think we need to add a shared precomp define, since those aren't needed for the FW side anyways - that could probably just be packed into the layout maps in some way.

The important thing, really, is defining which pins are "capable" of what; **"*almost* any"** implies there's still limitations in what can be mapped as analog for instance. My immediate (but not practiced) thought is to either pack into the upper byte of the board layout map, or having a unique map of bitmaps per board (not that I WANT a third map, but might be cleaner to read) of what features each pin is "capable" of - what I2C block it belongs to (if any), what half it functions as, and whether it supports analog.

alessandro-satanassi, 2025-04-17:
I agree, I think a third map in the Qt App section is the clearest thing.

SeongGino, 2025-05-20:
See #93 / TeamOpenFIRE/OpenFIRE-Boards@969ea6bd70c1f5e5b5777590b031331a21b45d66

closed 2025-05-20 by SeongGino