TeamOpenFIRE/OpenFIRE-App #81: [TESTING] OpenFIRE App Release v3.0 RC2 Test!

SeongGino, 2025-05-10:
## TL;DR: Download Releases: [For Windows 10+ (OpenFIREapp_win64.zip)](https://github.com/user-attachments/files/20140767/OpenFIREapp_win64.zip) | [For Linux (OpenFIRE_App-Linux-x86_64.zip)](https://github.com/user-attachments/files/20140786/OpenFIRE_App-Linux-x86_64.zip) [Updated @ 1:43 PM 05/10/2025]

This issue body won't have a lot of info, other than it took way too long to get to this point, and that the rest of this is a ***VERY ROUGH*** changelog* of the major features, additions, changes, and fixes since the last release version. Short and to the point, the version number bump and the wait should hopefully be very justified.

Consider this the (roughly) feature frozen version, barring any egregious UX issues. Even a perfectionist needs a point to put the pencil down and call a work "done" (for now).
###### *and I'm pretty sure I left out some still.

[image]

___
### Changelog for App v3.0 *"Tokinomiya"* (Release Candidate 2)
- Code Refactoring (like a lot...).
- App will now use the *Fusion* Qt style on Windows, which makes some elements of the UI more consistent with how they're intended to appear.
- App now builds for Qt6, which targets Windows 10-and-up.
  - Using this allows for the application to have proper dark-mode awareness.
- App will now seek for and update list of available boards in the background, removing the need for connecting a board prior to opening the application.
- Some interface elements will adjust colors to better suit either default Windows' light or dark theme.
- Elements like the tab bar now display icons for better clarity at a glance as to their functionality.
- Supported Boards and their default layouts can now be viewed in the app, from the *View Compatible Boards* option under the *Help* menu.
- A new fullscreen IR Emitter alignment assistant can now be viewed in the app, from the *Open IR Emitter Alignment Assistant* option under the *Help* menu.
  - Alignment Assistant is DPI-aware down to 480p and up to 4K, and will adjust for 16:9 or 4:3 displays.
- A new fullscreen calibration interface is used when initiating a profile calibration from the app.
- If attempt to get settings from a gun fails, the app will prompt the user if they want it to reset the board to bootloader to update the firmware.
- Submitting new settings to the microcontroller will attempt to verify each output, and retry up to three times if the response doesn't match what it sent.
- Version numbers with Git hashes will now be hyperlinked to the Git commit page on the main Firmware repository.

[image]

- Added Button Mapping Tab:
  - Allows for remapping functions of available buttons when aiming onscreen, offscreen, and when in gamepad mode.
  - Uses the new Descriptive Text model for dynamically showing tooltips when hovering over options.

- Board Layout Tab:
  - Pins layout uses a scroll area, being able to be navigated on smaller displays.
  - Mouse scrolling does not effect pin boxes anymore.
  - Custom Pin Layouts can now be exported to/imported from OpenFIRE Layout `.OFL` files
  - All of the available boards will show highlights when selecting certain pins.
  - App can support any amount of pins per board, depending on the board detected. This is to futureproof support for more/different kinds of microcontroller boards in the future.

- Gun Settings:
  - Settings uses a scroll area, being able to be navigated on smaller displays.
  - Detailed information that was previously displayed as delayed tooltips are now shown in the persistent description box at the bottom of the window, depending on the setting that's currently hovered.
  - The layout of settings has been reorganized to better fit their relationship to each other and certain pins.
  - Custom USB ID is now always shown in Hexadecimal, as it is displayed natively to the system, rather than the weird bodge of converting inputted Decimal to a small Hex code to the side.
  - Custom USB Product Name will now always enforce limit of up to 15 characters and only 8-bit character values.
  - I2C settings group has been added, allowing toggling and configuration of individual devices that may be connected to the Peripheral lines (if exists)
  - NeoPixels' invert static setting has been added.
  - Changes to Solenoid timing in FW 6.0 has been reflected, now providing basic "Solenoid On" and "Solenoid Autofire Off" timing settings, rather than a combination of Intervals and Autofire multiplier.

- Calibration Profiles:
  - App can support any amount of profiles from the gun, rather than being hardcoded to the default of 4.
  - Supports setting base aspect ratio of the display per cali profile.
  - Calibration buttons now summon the new fullscreen Calibration interface.
  - Descriptive texts added for the adjustable settings.

- Gun Tests:
  - Buttons tester area has been reorganized, making the buttons bigger.
  - Analog Stick, if available, now has a preview showing exact coordinates of the stick, as reported from the microcontroller.
  - Original (tiny) IR testing area has now been replaced with a fullscreen IR Test screen.
  - Microcontroller bootloader reset now uses the RP2040 magic baud number method rather than invoked by a command.

[image]

SeongGino, 2025-05-11:
Also, if you'd like to test on 64-bit ARM devices, we now have aarch64 builds working. These are built on Qt 6.7.3, which is the latest that can be run without GLIBC 2.83, which some distros like the default Raspberry Pi OS do not currently have. Aside from the default Wayland session not showing tab/button icons like it should, the rest of it functions on at least a Pi 3B, but would recommend something on the level of a base Pi 4 at least. We would appreciate it if distro developers could inform us whether this works on their builds or not.

[OpenFIREapp_aarch64.AppImage](https://github.com/SeongGino/OpenFIRE-App/releases/download/test/OpenFIRE_App-aarch64.AppImage)

SeongGino, 2025-05-12:
Builds in the OP and the aarch64 page have been updated. This was mostly a bunch of backend work fixing up the build pipeline, but there's a couple of user-facing changes:
 - Wayland components are now bundled with all Linux AppImages, so the App should show as a *native Wayland application* rather than running through XWayland.
 - Serial port is now cleared before grabbing the first batch of settings. This didn't originally seem to be an issue, but apparently the Raspberry Pi *in particular* was affected by this which always led to the first docking failing - perhaps due to the speed of my particular 3B board, it seems like it gets snagged on a temperature update code.
 - If the settings message received does not match the internal database of recognized settings types, then a timeout shouldn't erroneously abort the whole process anymore (which should also account for any rogue temp updates that gets sent in-between).

alessandro-satanassi, 2025-05-23:
S.O. Windows 11
Small problems in the Board Preview window:
the links related to "compatible with the" and "Any issues should be reported" do not automatically open the browser to the link.
Also when you go to the pin name, the longest one, the image moves because becoming the name in bold takes up more space, both with the left and right pins

These small problems persist even with the very latest commit 098722c

SeongGino, 2025-05-24:
The testing period has concluded. Thank you to everyone for your feedback (mainly on Discord).

closed 2025-05-24 by SeongGino